d-a-v/esp82xx-nonos-linklayer #8: General questions

martin-ger, 2017-12-19:
Looks interesting. Does it mean, I get a new implementation of the liblwip.a that links with Espressif blobs and the nonos/open-skd?

Any obvious reason, why IPv6 should not work? I am interested in testing/running the ESP8266 with IPv6...

d-a-v, 2017-12-19:
It is meant to replace liblwip.a. The proposed lwIP-v2 API is slightly different than v1's though.
And yes, there is absolutely no reason why IPv6 should not work.

martin-ger, 2017-12-20:
Thanks!

When trying to compile it with esp-open-sdk with " make -f Makefile.open install" it fails early with:

no rule for „/arch/cc.h“

I defined the 
PREFIX=/home/martin/github/esp-open-sdk/xtensa-lx106-elf
which points to my esp-open-sdk directory, so the binaries should be found, but I guess 

LWIP_INCLUDES_RELEASE=/home/martin/github/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip2

is wrong, as the is no lwip2 right now. Do I have to copy some files from the lwip2 into the sdk path?

Any help welcome.

d-a-v, 2017-12-20:
I made a pull-request for esp-open-sdk ([link](https://github.com/pfalcon/esp-open-sdk/pull/271)).
lwip2 is set there as a git-sub-module on a previous revision than current master here. Since that time the directory structure has changed and I did not yet update the relevant files.
You can use this pull-request for a try, until things are updated to match current master.
I think making lwip2 work with micropython will be the entrance ticket for the PR to be merged.

martin-ger, 2017-12-30:
Okay, thanks, can compile it. But enabling IPv6 is not so easy, as starting from cc.h there is obviously the assumption, that struct ip_addr is just a 32-bit val. Which part of the SW relies on that? GUess not the WiFi-driver but some upper layers - the espconn and the DHCP-server I guess?

d-a-v, 2017-12-30:
```struct ip_addr``` was in lwIP-1 and espressif copied it into the SDK.
It must not be anymore used under that form.
lwIP-v2 redefined the ip structures and propose helpers / macros  to deal with IPv4 and v6 addresses Check lwip-v2's ```ip_addr.h``` and their docs. I have little experience myself since I've been more concerned with bugs and stability in current implementations, even if IPv6 is something I'd like to try in the esp.
What I can say is that you must enable IPv6 in ```lwipopts.h``` and modify your environment to deal with both IPv4 and IPv6.

About DHCP server, the one provided by espressif is a direct update from the one provided by lwIP-1. Why not use the one inside lwIP-v2 which is already compatible with IPv6 ?

About espconn. This is done by espressif for espressif as an abstraction of the TCP/IP API. Trying to update it to IPv6 will require more work since you would also need to adapt the application that makes use of it. 

I started the job in arduino but it is far from finished. What I first did was to rename everywhere ```struct ip_addr``` to ```struct ip_addr_v4``` in both the ESP-SDK and the environment (esp8266/Arduino in that case) so we still have our IPv4 32bits address working like before where we need, especially when dealing with espressif blobs which will never know IPv6. Then I started to overload arduino's c++ ```IPAddress``` with new lwIP-2 structures. That could be sufficient since everytime an IP address is involved it is supposed to be held in ```IPAddress```. This is yet to be pursued.

andrelopesal, 2018-04-28:
Please, some way to make it work with [https://github.com/CHERTS/esp8266-devkit](url) ?

Thanks!

d-a-v, 2018-04-29:
@andrelopesal 
As said above, this a replacement of espressif's liblwip.a, it's up to this devkit project to use this library (if it is based on esp8266-nonos-sdk).

closed 2018-04-29 by d-a-v