d-a-v/esp82xx-nonos-linklayer #9: How to build lwip2 on Windows10?

TridentTD, 2018-01-02:
I can build lwip1.4 ( ESP8266-Arduino 2.4.0 )  on Windows system,
however when i try to build lwip2 ( ESP8266-Arduino 2.4.0 )  
It's occur error as the following.

How to build ?
Thank you.

```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder>make -f Makefile.arduino install
make -f makefiles/Makefile.build-lwip2 \
                        target=arduino \
                        SDK=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk \
                        LWIP_ESP2=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/include \
                        LWIP_ESP=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip/include \
                        LWIP_LIB=liblwip2.a \
                        LWIP_LIB_RELEASE=../../lib/liblwip2.a \
                        LWIP_INCLUDES_RELEASE=../include \
                        TOOLS=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                        TCP_MSS=536 BUILD=build-536 \
                        install
make[1]: Entering directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.glue-esp BUILD=build-536 TCP_MSS=536
make[2]: Entering directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -std=gnu99 -Wall -Wextra -c -Os -g -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=arduino -DTCP_MSS=536 -Ibuild-536 -I/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/include -I/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip/include -Iglue glue-esp/lwip-esp.c -o build-536/glue-esp/lwip-esp.o
glue-esp/lwip-esp.c: In function 'netif_add':
glue-esp/lwip-esp.c:481:4: error: #error
   #error
    ^
make[2]: *** [build-536/glue-esp/lwip-esp.o] Error 1
make[2]: Leaving directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make[1]: *** [liblwip2.a] Error 2
make[1]: Leaving directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make: *** [install] Error 2

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder>
```

d-a-v, 2018-01-02:
You probably have modified lwipopts.h and defined ```ENABLE_LOOPBACK``` to non-0.
If there is a reason for that, then comment the three lines around the #error.
Otherwise, re-checkout the configuration:
```
git checkout tools/sdk/lwip2/builder/glue-lwip/arduino/lwipopts.h
```

d-a-v, 2018-01-02:
You have modified the Makefiles, the LWIP_ESP2 I can see above is not part of the current Makefile.arduino.
Please report any error with original Makefiles or explain what you're trying to do.
```
cd tools/sdk/lwip2
mv builder builder.test # save your work
make install # restore original builder/ and compile with it
```

TridentTD, 2018-01-02:
I already mv builder folder to other name.
And make install   by original Makefile  at   sdk/lwip2  folder.
It's still occur error as the following .

```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>make install
git submodule update --init --recursive builder
error: pathspec 'builder' did not match any file(s) known to git.
make: *** [builder/lwip2-src/README] Error 1
```

d-a-v, 2018-01-02:
Yes sorry it should be
```
cd tools/sdk/lwip2
mv builder builder.test # save your work
git checkout builder
make install # restore original builder/ and compile with it
```

TridentTD, 2018-01-02:
Sorry for my poor English.
Still error ..

```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>git checkout builder
error: pathspec 'builder' did not match any file(s) known to git.

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>

```

d-a-v, 2018-01-02:
I have not tried the release yet, I'm always working with git.
lwip2 is the first submodule of esp8266/Arduino so maybe there is something wrong with the release.
I will check later, in the meantime you can
```
cd tools/sdk/lwip2
mv builder builder.test # save your work
git clone https://github.com/d-a-v/esp82xx-nonos-linklayer.git builder
make install # restore original builder/ and compile with it
```

TridentTD, 2018-01-02:
Still error ...

```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>git clone https://github.com/d-a-v/esp82xx-nonos-linklayer.git builder
Cloning into 'builder'...
remote: Counting objects: 436, done.
remote: Compressing objects: 100% (22/22), done.
Receiving objects: 100% (436/436), 288.07 KiB | 66.00 KiB/s, done.

Resolving deltas: 100% (225/225), done.

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>make install
git submodule update --init --recursive builder
error: pathspec 'builder' did not match any file(s) known to git.
make: *** [builder/lwip2-src/README] Error 1

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>
```

Problem is on  Makefile at    tools/sdk/lwip2/Makefile ?

```
all install clean: builder/lwip2-src/README
	make -C builder -f Makefile.arduino $@

download: builder/lwip2-src/README

builder/lwip2-src/README:
	git submodule update --init --recursive builder
```

d-a-v, 2018-01-02:
try:
cd builder
git submodule update --init --recursive
cd ..
make install

-- 
on mobile

TridentTD, 2018-01-03:
After git  & make install,
it still occurs error at the same place. 

> glue-esp/lwip-esp.c:481:4: error: #error

Total steps ..
```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>cd builder

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder>git submodule update --init --recursive
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'lwip2-src'
Cloning into 'C:/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder/lwip2-src'...
Submodule path 'lwip2-src': checked out '92f23d6ca0971a32f2085b9480e738d34174417b'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder>cd ..

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>make install
make -C builder -f Makefile.arduino install
make[1]: Entering directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.build-lwip2 \
                        target=arduino \
                        SDK=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk \
                        LWIP_ESP=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip/include \
                        LWIP_LIB=liblwip2.a \
                        LWIP_LIB_RELEASE=../../lib/liblwip2.a \
                        LWIP_INCLUDES_RELEASE=../include \
                        TOOLS=/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                        TCP_MSS=536 BUILD=build-536 \
                        install
make[2]: Entering directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
( \
        echo "// script-generated, extracted from espressif SDK's lwIP arch/cc.h"; \
        echo "#define LWIP_NO_STDINT_H 1"; \
        grep -e LWIP_ERR_T -e ^typedef /c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip/include/arch/cc.h \
        ) > glue-lwip/lwip-err-t.h
makefiles/make-lwip-hash
make -f makefiles/Makefile.glue-esp BUILD=build-536 TCP_MSS=536
make[3]: Entering directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -std=gnu99 -Wall -Wextra -c -Os -g -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=arduino -DTCP_MSS=536 -Ibuild-536 -I/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/include -I/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip/include -Iglue glue-esp/lwip-esp.c -o build-536/glue-esp/lwip-esp.o
glue-esp/lwip-esp.c: In function 'netif_add':
glue-esp/lwip-esp.c:481:4: error: #error
   #error
    ^
make[3]: *** [build-536/glue-esp/lwip-esp.o] Error 1
make[3]: Leaving directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make[2]: *** [liblwip2.a] Error 2
make[2]: Leaving directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make[1]: *** [install] Error 2
make[1]: Leaving directory `/c/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/builder'
make: *** [install] Error 2

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>
```

d-a-v, 2018-01-07:
I don't understand why you would have this error. I need to check on a windows computer.
What you can try is:
```
cd tools/sdk/lwip2
mv builder builder.unwanted
git clone --recursive https://github.com/d-a-v/esp82xx-nonos-linklayer
make install
```

TridentTD, 2018-01-07:
Thank  you .

I have just tried, the result as following..

```
C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>git clone --recursive https://github.com/d-a-v/esp82xx-nonos-linklayer
Cloning into 'esp82xx-nonos-linklayer'...
remote: Counting objects: 440, done.
remote: Compressing objects: 100% (25/25), done.
remote: Total 440 (delta 8), reused 25 (delta 7), pack-reused 405
Receiving objects: 100% (440/440), 293.88 KiB | 367.00 KiB/s, done.
Resolving deltas: 100% (226/226), done.
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'lwip2-src'
Cloning into 'C:/Users/Superman/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/sdk/lwip2/esp82xx-nonos-linklayer/lwip2-src'...
remote: Counting objects: 47169, done.
remote: Compressing objects: 100% (11163/11163), done.
remote: Total 47169 (delta 35520), reused 47056 (delta 35449)
Receiving objects: 100% (47169/47169), 8.87 MiB | 385.00 KiB/s, done.
Resolving deltas: 100% (35520/35520), done.
Submodule path 'lwip2-src': checked out '92f23d6ca0971a32f2085b9480e738d34174417b'

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>make install
git submodule update --init --recursive builder
error: pathspec 'builder' did not match any file(s) known to git.
make: *** [builder/lwip2-src/README] Error 1

C:\Users\Superman\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2>
```

TridentTD, 2018-01-07:
Ok , finally  I  mv "esp82xx-nonos-linklayer" folder to "builder"
and can  compile lwip2.

Thank you very much.

d-a-v, 2018-01-08:
Good, sorry for the "builder" directory name, I updated my previous comment :confused:

closed 2018-01-08 by d-a-v