d-a-v/esp82xx-nonos-linklayer #15: Compiling with PPP

andrelopesal, 2018-04-23:
First, thank you for your work!

I'm trying to compile with PPP, so I added the PPP defines in the "lwipopts.h" file in glue-lwip->arduino, I added "$(patsubst %.c,$(BUILD)/%.o,$(wildcard netif/ppp/*.c)) \","$(patsubst %.c,$(BUILD)/%.o,$(wildcard netif/ppp/polarssl/*.c)) \", "-include $(BUILD)/*/*/*/*/*.d" and "-include $(BUILD)/*/*/*/*/*/*.d" to the Makefile.lwip2 file. But when I use liblwip2.a in the project, I get the Undefined reference for PPP functions, any ideia?

d-a-v, 2018-04-25:
Do you `make clean` / `make install` from arduino's `tools/sdk/lwip2/` directory ?
In that case, can you see whether PPP is compiling in the logs ?

andrelopesal, 2018-04-25:
The problem was extern "C" {} on ppp includes in my project. The challenge
now is to do NAT between PPP and wifi station.

d-a-v, 2018-04-26:
NAT is coming (it's waiting for fixes in a branch).

andrelopesal, 2018-04-26:
Thanks so much, meanwhile I'll try this: [https://github.com/ajaybhargav/lwip_nat](url).