d-a-v/esp82xx-nonos-linklayer #27: Open SDK

someburner, 2019-02-09:
Looks like this is coming along! I'm interested in trying it out with my project but using esp-open-sdk.

I was able to get it to build with a fresh clone of the PR to esp-open-sdk, and then incrementally applied changes, but I got stuck at around this point: ea83a83fed8867c4bc9bbc01cbfc8d5ce240bb3b

From there on I can't get it to compile, getting the message that I should be using "lwip 1.4 headers" - I tried taking them from Arduino but no luck, I guess I don't have a good enough grasp on which headers should go where? I also tried compiling with what I had, but couldn't figure out what had to be done for `ipv4_addr`.

If it's not a priority I don't mind working on it- I just need some more direction on the headers and what sorts of things I might need to change in my project to resolve `ipv4_addr` vs `ip_addr`. Thanks in advance!

d-a-v, 2019-02-10:
`ip_addr_t` is lwIP IP address structure.
in lwIP v2, this structure evolves when IPv6 is enabled.
nonos-sdk fw will ever only knows IPv4 with the initial structure, we must use this one when we talk to fw.
So I had to rename all SDK's references to `ip_addr_t` to use `ipv4_addr_t` instead because this name `ipv4_addr_t` does not exist in lwIP (I tried several ways, including using lwIP's `ip4_addr_t`. In the end `ipv4_addr_t` was the easiest way).

When IPv6 is not enabled, `ipv4_addr_t` is exactly the same as `ip_addr_t` (also as `ipv4_addr` and `ip_addr`).

someburner, 2019-02-11:
Okay, and what about the error in `arch/cc.h` about "lwip 1.4"? I understand the reason for why lwip with ipv6 requires new structs, but since this build is using relative paths everywhere it's very difficult to figure out where header files are being sourced from.

d-a-v, 2019-02-11:
I remember I had to replace a `#include "something"` by `#include <samething>` in sdk,
that, so `-I` is considered, otherwise same dir is used.

Tell me more about the error you have.

someburner, 2019-02-11:
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/a501b5722bd050f41a9351e69ca75937150ec138/glue-lwip/arduino/lwipopts.h#L3579

d-a-v, 2019-02-11:
That's because espressif changed this type a year ago.
This [script](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/master/makefiles/Makefile.build-lwip2#L26) extracts it from SDK.
You can use `-DLWIP_ERR_T=s8` (or whatever `lwip_err_t` is in you sdk version).

someburner, 2019-02-11:
Okay thanks- still I am wondering which version/release is being used exactly? or should it not matter?

d-a-v, 2019-02-11:
On this repository, master is always better, generally.
The version used in arduino (which is supposed to be stable) is linked from [there:builder/](https://github.com/esp8266/Arduino/tree/master/tools/sdk/lwip2) as a submodule.

someburner, 2019-02-13:
Okay so here's where I'm at. I redid all the steps I took before, but also took care to build an esp-open-sdk using the sample libmain version mentioned in `glue-esp/lwip-esp.c` which is `sdk-2.2.1(cfd48f3)`.

My [opensdk](https://github.com/someburner/esp82xx-nonos-linklayer/tree/opensdk) branch is your master up to (but not including) 6bb92aa727befdaaa5c1fd498493a46c2858d625 which is where the ipv6 is introduced and I can no longer build with esp-open-sdk.

I also made [open-sdk-broken](https://github.com/someburner/esp82xx-nonos-linklayer/tree/open-sdk-broken) which is up to your master and includes how I would merge those changes for the makefile loop with ipv4/6 and mss 536/1460. This does not compile though- getting errors like:

```
make -f makefiles/Makefile.glue-esp BUILD=build-536-v4
make[3]: Entering directory '/home/jeff/build/esp8266/sdk2/lwip2'
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -std=gnu99 -Wall -Wextra -c -Os -g -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=open -DTCP_MSS=536 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -Ibuild-536-v4 -I/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include -I/home/jeff/build/esp8266/sdk2/lwip2/lwip14-headers -Iglue glue-esp/lwip-esp.c -o build-536-v4/glue-esp/lwip-esp.o
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar cru liblwip2-536.a build-536-v4/glue-esp/lwip-esp.o
make[3]: Leaving directory '/home/jeff/build/esp8266/sdk2/lwip2'
make -f makefiles/Makefile.glue target=open BUILD=build-536-v4 TCP_MSS=536 LWIP_FEATURES=0 LWIP_IPV6=0
make[3]: Entering directory '/home/jeff/build/esp8266/sdk2/lwip2'
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -std=gnu99 -Wall -Wextra -c -Os -g -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=open -DTCP_MSS=536 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -Ibuild-536-v4 -Iglue -Iglue-lwip -Iglue-lwip/open -Ilwip2-src/src/include -I/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include glue/uprint.c -o build-536-v4/glue/uprint.o
In file included from glue/glue.h:41:0,
                 from glue/uprint.h:7,
                 from glue/uprint.c:31:
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:353:20: error: field 'ip' has incomplete type
     struct ip_addr ip;
                    ^~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:358:20: error: field 'start_ip' has incomplete type
     struct ip_addr start_ip;
                    ^~~~~~~~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:359:20: error: field 'end_ip' has incomplete type
     struct ip_addr end_ip;
                    ^~~~~~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:506:20: error: field 'ip' has incomplete type
     struct ip_addr ip;
                    ^~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:507:20: error: field 'mask' has incomplete type
     struct ip_addr mask;
                    ^~~~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:508:20: error: field 'gw' has incomplete type
     struct ip_addr gw;
                    ^~
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/user_interface.h:518:20: error: field 'ip' has incomplete type
     struct ip_addr ip;
                    ^~
make[3]: *** [makefiles/Makefile.rules:8: build-536-v4/glue/uprint.o] Error 1
```

I commented out `ip_addr` in `user_interface.h` and it compiles, but that just shows there is something not right with the headers. Linking against a project results in lots of link errors, particularly with sntp. There is also the issue with `sys/pgmspace.h` being arduino-specific.

To me it seems the espconn port steered towards Arduino pretty heavily. If I were to go further I think I'd have to start at the point that still compiles for open-sdk- but maybe not? It's hard for me to decipher the Arduino-specific portions.

someburner, 2019-02-14:
Deleted previous comment since it seems those arose from trying to start with a project that had too much going on dependency-wise. Now where I'm at:

Re-compiled esp-open-sdk from the PR. Was getting linker errors for `free` and `calloc` like this:

```
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip2.a(mem.o): In function `mem_malloc':
/home/jeff/build/esp8266/sdk2/lwip2/lwip2-src/src/core/mem.c:136: undefined reference to `free'
/home/jeff/build/esp8266/sdk2/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip2.a(mem.o): In function `mem_free':
/home/jeff/build/esp8266/sdk2/lwip2/lwip2-src/src/core/mem.c:151: undefined reference to `free'
collect2: error: ld returned 1 exit status
```

I think it's looking for those in irom so I added `malloc` + `free` methods in flash in my project that call `os_xxx` methods. I was previously getting that to link some other way and it was crashing things immediately.

Next was fixing includes for my project. I think the issue is that esp-open-sdk copies `esp-open-lwip` includes into `sysroot`, so I needed to include the `lwip2` folder in CFLAGS.

Now my test project can boot, and including `<lwip/init.h>` gets MAJOR==2. But it seems something is broken still: Calling `wifi_station_dhcpc_start()` shows the message that dhcp is starting up, and then calling `wifi_station_connect()` I get the `EVENT_STAMODE_CONNECTED` event. However after that, no IP address. I also tried setting a static IP/GW/mask and then calling connect, but I get:

```
connected with TestWiFi, channel 11
ip:192.168.1.55,mask:255.255.255.0,gw:192.168.1.1
lwESP: netif_set_up is called??ip:192.168.1.55,mask:255.255.255.0,11Wifi connected to ssid TestWiFi, ch 11
Wifi got ip:192.168.1.55,mask:255.255.255.0,gw:192.168.1.1
```

The router shows the ESP as associated but no IP. Is this a symptom of the older version in use? Or should I be getting an IP with dhcp? Or maybe something else I'm missing?

someburner, 2019-02-14:
okay I got it all working up to 163bb82dd97e985df76c07d1c3f7325212cbdd5e. Still need to figure out the deal with linking for `free` and `malloc`, but overall I'm pretty happy and look forward to testing this on a bigger scale. If there's any interest I can list out my steps for open-sdk.

d-a-v, 2019-02-15:
> Doing just the bare minimum to switch to lwip2 and now no more disconnects. Thanks @d-a-v !

:+1: Thanks to you!

Is there any modification you had to do in my repository ?

> If there's any interest I can list out my steps for open-sdk.

There is. You may start posting your steps in a new issue in esp-open-sdk.
(It will replace unifinished In https://github.com/pfalcon/esp-open-sdk/pull/271)

andrethomas, 2019-02-15:
Definitely some interest from my side also...

someburner, 2019-02-15:
Okay- I would like to maintain compatibility with the work in this repo, and right now it is forked off from a fairly early commit that introduced all the changes to put debugs into PROGMEM. The gluedebug.h had a lot going on so I just kept what it had originally. My project places all strings into flash NodeMCU-style already so for me it wasn't a problem. Maybe it's not actually that much that needs to be changed to make it compatible with both, but the commit I'm referring to is https://github.com/d-a-v/esp82xx-nonos-linklayer/commit/d45ac2aa68f7361556935fe8dbcfba94381c4e58.

My forked version is [here](https://github.com/someburner/esp82xx-nonos-linklayer). If you scroll through commit history on the `open` branch you can see the point where the commits by @d-a-v start turning into commits I cherry-picked.

I also have my own version of `esp-open-sdk` that uses recent GCC/newer crosstool-ng, and I made a branch that should compile without issue on most linux distros.

```sh
git clone --recursive https://github.com/someburner/esp-open-sdk.git -b sdk2-lwip2
```

That said- the later commits on my `esp82xx-nonos-linklayer` fork added some additional patches intended to eliminate any manual patching and should work with the original open-sdk and whatever SDK it's currently using.

So I think the best course of action would be to try to resolve the strings-in-flash thing, and then after that a diff between the 2 repos should show all the other changes I made (which are not that many, and mostly self-describing). I also want to try to get up to lwip 2.1.2.

I have a feeling pfalcon is going to want sntp implementation to be provided one way or the other.

someburner, 2019-02-16:
Okay I finished getting it up to master of this repo. Like I said before, I generally see an improvement on the ~5 or 6 devices I've ran this on. However I have noticed one thing that I do not see with vendor lwip1.4 - which is resets due to exceptions. I am almost positive this must be coming from the new lwip (base fw is stable running on >1K devices - no exceptions ever, all restarts/low-mem are handled gracefully). I am using TCP_MSS = 1460.

This might warrant a separate issue but wanted to get some feedback on stability at 1460 first.

someburner, 2019-02-21:
@d-a-v - Think I may have found the issue:

When calling `espconn_disconnect` in my project, the registered disconnect callback never gets called. I put debugs in `glue-lwip/espconn_tcp.c` and found that `espconn_Task` is never being entered. It should be though- at the end of `espconn_client_close` there is

```c
ets_post(espconn_TaskPrio, SIG_ESPCONN_CLOSE, (uint32_t)pclose);
```

Seems that `espconn_init` is never called. Is this on purpose? Calling `espconn_init();` in my project on init seems to work, and the tasks work after that.

d-a-v, 2019-02-22:
First, sorry for being silent to your questions,
thanks for your work,

> I had a question- what is the deal with TCP_MSS 536 vs. 1460? Is 1460 considered stable or what is wrong with it?

Only ram occupation. Also, working with 536 allowed me to check on good quality code that does not rely on this 1460 constant (that is on arduino core side). It is safe to work with 1460, allows better bandwidth (less inter-packet latency).

> Seems that espconn_init is never called.

I have never used espconn (Arduino core does not use it). I have only ported it and included in the library for further test, also because it was requested at some point. I have never had any feedback.
Feel free to fix anything related with espconn with that in mind.
I think this espconn was (from lwIP v1.4 times) initially a rename of lwIP's `netconn`. I don't know why this rename or what are the additions / differences.

About my rather complex debug print system. That was long ago and at that time I couldn't configure my usb/serial to 74880bauds. That is why I am using buffers. I guess this could be completely removed.

> So I think the best course of action would be to try to resolve the strings-in-flash thing

What are the issues ?

someburner, 2019-02-22:
> sorry for being silent to your questions

Not a problem! But I do hope to get my changes merged in while this is still fresh in my head.

> Only ram occupation.

Okay good to know. On a related note, there were other changes in `lwipopts.h` that diverge from Espressif lwip14 `lwipopts.h`, like `MEM_SIZE`. Some are commented clearly as required for ESP8266, but would be nice to have more comments explaining the changes. I am barely acquainted with lwip at all though so I have mostly copied the Arduino changes for open-sdk.

> I have never used espconn (Arduino core does not use it). I have only ported it and included in the library for further test, also because it was requested at some point. I have never had any feedback.
Feel free to fix anything related with espconn with that in mind.

Okay that makes much more sense. After fixing `espconn_init` it appears to function the same as the SDK/lwip14 espconn. I've had this going a couple days now on my toughest devices and 20 others with good connections and have not seen an exception yet. Hopefully that was indeed the issue, but I can't say for sure for a few more days at least.

> I guess this could be completely removed.

In that case I will try to resolve the issue by using some `#ifdef`s for Arduino vs. other, which is already necessary for SNTP implementation differences. I'll fork and submit a PR soon.

someburner, 2019-02-22:
For the record, here is what happens when trying to link after commit d45ac2aa68f7361556935fe8dbcfba94381c4e58:

```
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-abort.o):(.literal+0x0): undefined reference to `_exit'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-abort.o): in function `abort':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdlib/abort.c:63: undefined reference to `_exit'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-fflush.o):(.literal+0x0): undefined reference to `_free_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-fflush.o): in function `__sflush_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/fflush.c:162: undefined reference to `_free_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-findfp.o):(.literal+0x4): undefined reference to `_malloc_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-findfp.o): in function `__sfmoreglue':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/findfp.c:88: undefined reference to `_malloc_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-signal.o):(.literal+0x0): undefined reference to `_getpid_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-signal.o):(.literal+0x4): undefined reference to `_kill_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-signal.o): in function `_init_signal_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:110: undefined reference to `_malloc_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-signal.o): in function `_raise_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:163: undefined reference to `_getpid_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/signal/signal.c:163: undefined reference to `_kill_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o):(.literal+0x0): undefined reference to `_read_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o):(.literal+0x4): undefined reference to `_lseek_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o):(.literal+0x8): undefined reference to `_write_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o):(.literal+0xc): undefined reference to `_close_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o): in function `__sread':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/stdio.c:48: undefined reference to `_read_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o): in function `__swrite':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/stdio.c:89: undefined reference to `_lseek_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/stdio.c:97: undefined reference to `_write_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/jeff/build/esp8266/sdk2.2.2/sdk/lib/libc.a(lib_a-stdio.o): in function `__sseek':
[... cut ...]
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/fvwrite.c:154: undefined reference to `_realloc_r'
/home/jeff/build/esp8266/sdk2.2.2/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/7.4.0/../../../../xtensa-lx106-elf/bin/ld: /home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/fvwrite.c:160: undefined reference to `_free_r'
```

d-a-v, 2019-02-23:
Nice work. What is preventing you to use master ?
Is it only because of ipv4_addr_t ?

someburner, 2019-02-23:
No I'm already on master with my `open-dev` branch, but they are cherry-picked commits with some edits so I'm trying to patch my changes onto your current master.

d-a-v, 2019-07-16:
@someburner latest "locking" commit is good for everyone.

someburner, 2019-07-16:
@d-a-v Thanks. I will pull that in on my esp-open-sdk.