d-a-v/esp82xx-nonos-linklayer #36: Stub implementation

benpye, 2019-08-12:
Hey,

I'm really happy I've stumbled on this project. I want to use an ESP8266 to send/receive raw ethernet frames. There's currently the esp-raw project, it hacks about with lwip's netconn object to hook functions and works, but is obviously not perfect.

I have been wondering what it would take to stub out the lwip calls from the wifi stack so that we can just send/receive our own ethernet frames and ignore everything else. This is achievable with the RTOS SDK however I'd like to try and use my device as an SDIO slave which is not supported with the RTOS SDK.

d-a-v, 2019-08-15:
With this project and arduino core,
* you can send raw ethernet frames using [`new_linkoutput()`==`netif->linkoutput()`](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/master/glue-lwip/lwip-git.c#L207). You'll have to put your data into an lwIP's pbuf that will be released by the wifi firmware once sent
* you can intercept every received packet using `phy_capture` function like in this example: [netdump](https://github.com/d-a-v/EspGoodies/blob/master/examples/netdump/netdump.ino#L24),
or by using lwIP's hooks for unmanaged protocols

benpye, 2019-08-17:
@d-a-v whilst that is certainly an option I suppose it isn't my ideal state. I would like to - if possible - eliminate lwip entirely. Part of this excercise is an attempt to minimize the amount of unsafe code running in my project - I plan to use the ESP8266 forwarding raw ethernet frames - but then use smoltcp on another (ARM) MCU where I can use Rust. It's for that reason I would like to try and get a stub implementation, sufficient to satisfy the WiFi blobs but no real functionality beyond that.