d-a-v/esp82xx-nonos-linklayer #38: Thoughts on RTOS_SDK options

someburner, 2019-09-06:
Scanning over the [RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK) I saw a couple changes they've made to their lwip. Thoughts on these?

https://github.com/espressif/ESP8266_RTOS_SDK/commit/d64f04bc95845d7de3d8cfbfe885ead1b80f10da

https://github.com/someburner/esp82xx-nonos-linklayer-opensdk/commit/893a240de8e551935edc40a7b5ca60bd79d89170

Second one is on my repo, that's extracted from their kconfig defaults.

There are others too. Just wondering if you think it's worth following their changes for certain lwip things.

d-a-v, 2019-09-06:
Yes you can add these here too, or is this a feature request ?

Let me claim also here that espressif is again doing things not the best with lwIP:
Instead of having a patch policy, they integrate their changes in their local copy. Once again they shoot themselves in the foot because it makes lwIP upgrade much more difficult (well, much less than with nonos-sdk though :)

someburner, 2019-09-06:
If I can resolve the open-sdk PR for my project I can make these changes and submit another PR.

I agree they really did not think this through at all. Maintaining a set of patches is much cleaner and easier for other developers to follow.

someburner, 2022-12-07:
Another one.. scanning over NONOS_SDK commits from last year.. https://github.com/espressif/ESP8266_NONOS_SDK/commit/a9a39bbc9b838de50d60f2448ecee6080bb332e4

I wish there was some sort of description for these regarding the set of circumstances leading them to make these changes.