d-a-v/esp82xx-nonos-linklayer #40: LWIP submodule

TD-er, 2019-12-06:
I am not entirely sure where PlatformIO does get the link to the git repo of LWIP.
It looks like it is defined here:
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/master/.gitmodules

Would it help if a clone of the repo is made on Github which is then pointed to by this definition?

The problem is, none of my systems can build ESP stuff anymore as PIO really refuses to do anything because the savannah.nongnu server reports bad gateway when accessed from command line git.
It does load fine in the browser though, but that's not really helpful for PIO.

d-a-v, 2019-12-07:
This is solved by the latest pushes.
This however won't solve esp-easy-mega configurations which are using previous version of esp8266-arduino core: proper CI scripts will have to be used to not clone the sub sub module.

TD-er, 2019-12-07:
Do you have a link of how to clone these?

d-a-v, 2019-12-07:
https://github.com/esp8266/Arduino/pull/6887

d-a-v, 2022-01-03:
@TD-er can this be closed ?
FYI, upstream git reference is [there](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/450bb63c1bc8b35770ca7f246945cf383569f52f/makefiles/Makefile.defs#L2) and [there](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/27f9894f202119c5a611fc1d2691abb810f77653/CMakeLists.txt#L10).