d-a-v/esp82xx-nonos-linklayer #41: Could we add some build instructions?

rsaxvc, 2020-01-09:
Could we please add some instructions for building from source?

Here's what I did, though it seems quite convoluted and probably not correct

1. Install latest arduino IDE
2. Install ESP8266 core 2.6.3
3. Close arduino IDE and remove 2.6.3 folder
4. Git clone ESP8266 core and checkout 2.6.3 tag
5. Hardcode tools path in Makefile.arduino to TOOLS=~/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-
6. Replace TOOLS=../../$(TOOLS) with TOOLS=$(TOOLS) in Makefile.build-lwip2
7. Run make install

d-a-v, 2020-01-09:
I would use the official documentation for installing the core:
1. https://arduino-esp8266.readthedocs.io/en/stable/installing.html#using-git-version
    (the mentioned `./get.py` script and `git submodule` command will download everything which is needed)

Then the included makefile should work:

2. `cd <esp8266-arduino-root-location>; cd tools/sdk/lwip2/`
3. `cat README.md Makefile`
4. `make install`

rsaxvc, 2020-01-21:
Thank you, this worked much better than my attempt.

closed 2020-01-21 by rsaxvc