d-a-v/esp82xx-nonos-linklayer #43: `__attribute((weak))` & `extern` usage in headers

mcspr, 2020-01-31:
- There is possible issue with
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7aade536beba9b3d2174a81cbb498e871c39be89/glue-lwip/open/lwipopts.h#L3592
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7aade536beba9b3d2174a81cbb498e871c39be89/glue-lwip/arduino/lwipopts.h#L3615
As noticed here:
https://github.com/bxparks/UnixHostDuino/commit/9c0755d303d66c4a30cd35589efa9fdb9c9360c5#r36778464
`weak` should be in `.c` instead of header, otherwise there is ordering issue when both compilation units include lwip headers and define the function

- (https://github.com/d-a-v/esp82xx-nonos-linklayer/commit/e23a07e5f9ccb43d36777c497fc34d044e3ecf3b) Referencing the PR to esp8266 Core that added these two weak functions:
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7aade536beba9b3d2174a81cbb498e871c39be89/glue-lwip/arduino/lwipopts.h#L3577
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7aade536beba9b3d2174a81cbb498e871c39be89/glue-lwip/arduino/lwipopts.h#L3595

Does it even need `extern` part? Quoting https://en.wikipedia.org/wiki/External_variable#Definition,_declaration_and_the_extern_keyword
> The extern keyword applied to a function prototype does absolutely nothing (the extern keyword applied to a function definition is, of course, non-sensical). A function prototype is always a declaration and never a definition. Also, in standard C, a function is always external, but some compiler extensions allow a function to be defined inside a function.

d-a-v, 2020-01-31:
That is right, the `weak` attribute should be in `.c`.
You are also right about `extern` being useless.

Would you like to make a PR right here ?
I would then include it in https://github.com/esp8266/Arduino/pull/6887

mcspr, 2020-01-31:
Sure, will try to do it today

PR #44 (2020-01-31): Headers fix
fix #43
closed 2020-01-31 by d-a-v