d-a-v/esp82xx-nonos-linklayer #49: Enabling `LWIP_DEBUG` breaks compilation

Are52Tap, 2021-05-11:
Putting `#define LWIP_DEBUG 1` in `glue-lwip/arduino/lwipopts.h` breaks compilation when doing `make` in ...`tools/sdk/lwip2` of [esp8266](https://github.com/esp8266/Arduino):

```
nathan@casparu:/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2$ make
git submodule update --init --recursive builder
make -C builder -f Makefile.arduino all
make[1]: Entering directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
for v6 in 0 1; do \
for mss in 536 1460; do \
for feat in 0 1; do \
        test \( $feat = 0 \) -a \( $v6 = 1 \) && continue; \
        featname= ;\
        test $feat = 1 && featname=-feat ;\
        make -f makefiles/Makefile.build-lwip2 \
                target=arduino \
                DEFINE_TARGET=ARDUINO \
                SDK=../.. \
                LWIP_ESP=glue-esp/lwip-1.4-arduino/include \
                LWIP_LIB=liblwip$((2+4*v6))-$mss$featname.a \
                LWIP_LIB_RELEASE=../../lib/liblwip$((2+4*v6))-$mss$featname.a \
                LWIP_INCLUDES_RELEASE=../include \
                TOOLS=/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf- \
                TCP_MSS=$mss LWIP_FEATURES=$feat LWIP_IPV6=$v6 \
                BUILD=build-$mss$featname-v$((4+2*v6)) \
                all ; \
done; done; done
make[2]: Entering directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
git clone --depth=1 -b STABLE-2_1_2_RELEASE https://git.savannah.nongnu.org/git/lwip.git lwip2-src
Cloning into 'lwip2-src'...
remote: Counting objects: 438, done.
remote: Compressing objects: 100% (389/389), done.
remote: Total 438 (delta 67), reused 163 (delta 28)
Receiving objects: 100% (438/438), 1.29 MiB | 963.00 KiB/s, done.
Resolving deltas: 100% (67/67), done.
Note: switching to '159e31b689577dbf69cf0683bbaffbd71fa5ee10'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

( \
echo "// script-generated, extracted from espressif SDK's lwIP arch/cc.h"; \
echo "#define LWIP_NO_STDINT_H 1"; \
grep -e LWIP_ERR_T -e ^typedef glue-esp/lwip-1.4-arduino/include/arch/cc.h \
) > glue-lwip/lwip-err-t.h
makefiles/make-lwip-hash
make -f makefiles/Makefile.glue-esp BUILD=build-536-v4 V=0
make[3]: Entering directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
C  build-536-v4/glue-esp/lwip-esp.o
AR liblwip2-536.a
make[3]: Leaving directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.glue target=arduino BUILD=build-536-v4 TCP_MSS=536 LWIP_FEATURES=0 LWIP_IPV6=0 V=0
make[3]: Entering directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
C  build-536-v4/glue/uprint.o
C  build-536-v4/glue/doprint.o
C  build-536-v4/glue-lwip/lwip-git.o
glue-lwip/lwip-git.c:531:9: note: '#pragma message: 

-------- TCP_MSS = 536 --------
[... 289 lines cut ...]
  148 |       LWIP_DEBUGF( PING_DEBUG, (" %"U32_F" ms\n", (sys_now()-ping_time)));
      |                                 ^~~~
lwip2-src/src/include/lwip/arch.h:81:42: note: in definition of macro 'LWIP_PLATFORM_DIAG'
   81 | #define LWIP_PLATFORM_DIAG(x) do {printf x;} while(0)
      |                                          ^
glue-lwip/esp-ping.c:148:7: note: in expansion of macro 'LWIP_DEBUGF'
  148 |       LWIP_DEBUGF( PING_DEBUG, (" %"U32_F" ms\n", (sys_now()-ping_time)));
      |       ^~~~~~~~~~~
glue-lwip/esp-ping.c:148:35: note: format string is defined here
  148 |       LWIP_DEBUGF( PING_DEBUG, (" %"U32_F" ms\n", (sys_now()-ping_time)));
      |                                   ^
make[3]: *** [makefiles/Makefile.rules:8: build-1460-feat-v6/glue-lwip/esp-ping.o] Error 1
make[3]: Leaving directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make[2]: *** [makefiles/Makefile.build-lwip2:53: liblwip6-1460-feat.a] Error 2
make[2]: Leaving directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make[1]: *** [Makefile.arduino:4: all] Error 2
make[1]: Leaving directory '/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make: *** [Makefile:3: all] Error 2
```

I've tried a couple things... e.g. changing `LWIP_NO_INTTYPES_H` above the `U32_F` (and etc.) definitions to `1` did not fix the problem. `U32_F` is not the only problem.

I am not certain if enabling it there is the recommended place (thought it was okay based on [this](https://lwip.fandom.com/wiki/Debugging_lwIP).

Here is the patch `git diff > lwipdebug.patch`: [lwipdebug.txt](https://github.com/d-a-v/esp82xx-nonos-linklayer/files/6457638/lwipdebug.txt)

Are52Tap, 2021-05-12:
@d-a-v So that expansion error is still persisting actually (as compiled with the modifications as mentioned in my previous UPDATE). 

Here `glue-lwip/esp-ping.c` is compiled (as previously mentioned) and throws that expansion error:
```
$ xtensa-lx106-elf-gcc -std=gnu99 -H -Wall -Wextra -c -Os -g -free -fipa-pta -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -DLWIP_BUILD -DUSE_OPTIMIZE_PRINTF -DTARGET=arduino -DARDUINO=1 -DTCP_MSS=536 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -Ibuild-536-v4 -Iglue -Iglue-lwip -Iglue-lwip/arduino -Ilwip2-src/src/include -I../../include glue-lwip/esp-ping.c -o build-536-v4/glue-lwip/esp-ping.o
. lwip2-src/src/include/lwip/opt.h
.. glue-lwip/arduino/lwipopts.h
... glue/gluedebug.h
.... ../../include/osapi.h
..... build-536-v4/user_config.h
... lwip2-src/src/include/lwip/debug.h
.... lwip2-src/src/include/lwip/arch.h
..... glue-lwip/arch/cc.h
...... glue-lwip/lwip-err-t.h
...... ../../include/ets_sys.h
....... ../../include/c_types.h
....... ../../include/eagle_soc.h
...... glue/esp-missing.h
....... ../../include/ipv4_addr.h
........ lwip2-src/src/include/lwip/init.h
......... lwip2-src/src/include/lwip/opt.h
........ lwip2-src/src/include/lwip/ip_addr.h
......... lwip2-src/src/include/lwip/def.h
.......... lwip2-src/src/include/lwip/arch.h
......... lwip2-src/src/include/lwip/ip4_addr.h
......... lwip2-src/src/include/lwip/ip6_addr.h
.......... lwip2-src/src/include/lwip/def.h
...... ../../include/mem.h
...... glue/glue.h
....... glue/gluedebug.h
....... ../../include/user_interface.h
........ ../../include/os_type.h
......... ../../include/ets_sys.h
........ ../../include/ipv4_addr.h
........ ../../include/queue.h
........ build-536-v4/user_config.h
........ ../../include/spi_flash.h
......... ../../include/spi_flash_geometry.h
........ ../../include/gpio.h
... glue-lwip/lwip-git-hash.h
. lwip2-src/src/include/lwip/mem.h
. lwip2-src/src/include/lwip/raw.h
.. lwip2-src/src/include/lwip/pbuf.h
... lwip2-src/src/include/lwip/err.h
.. lwip2-src/src/include/lwip/ip.h
... lwip2-src/src/include/lwip/netif.h
.... lwip2-src/src/include/lwip/stats.h
..... lwip2-src/src/include/lwip/memp.h
...... lwip2-src/src/include/lwip/priv/memp_std.h
...... lwip2-src/src/include/lwip/priv/memp_std.h
...... lwip2-src/src/include/lwip/priv/memp_priv.h
....... lwip2-src/src/include/lwip/priv/mem_priv.h
...... lwip2-src/src/include/lwip/stats.h
... lwip2-src/src/include/lwip/ip4.h
.... lwip2-src/src/include/lwip/prot/ip4.h
... lwip2-src/src/include/lwip/ip6.h
... lwip2-src/src/include/lwip/prot/ip.h
. lwip2-src/src/include/lwip/icmp.h
.. lwip2-src/src/include/lwip/prot/icmp.h
. lwip2-src/src/include/lwip/sys.h
. lwip2-src/src/include/lwip/timeouts.h
. lwip2-src/src/include/lwip/inet_chksum.h
. ../../include/os_type.h
. glue-lwip/lwip/apps-esp/ping.h
In file included from glue-lwip/arduino/lwipopts.h:55,
                 from lwip2-src/src/include/lwip/opt.h:51,
                 from glue-lwip/esp-ping.c:45:
glue-lwip/esp-ping.c: In function 'ping_recv':
lwip2-src/src/include/lwip/arch.h:162:15: error: expected ',' or ';' before '__INT32'
  162 | #define U32_F PRIu32
[... 29 lines cut ...]
  148 |       LWIP_DEBUGF( PING_DEBUG, (" %"U32_F" ms\n", (sys_now()-ping_time)));
      |       ^~~~~~~~~~~
glue/gluedebug.h:84:17: warning: too many arguments for format [-Wformat-extra-args]
   84 |   doprint_minus(flash_str, ##__VA_ARGS__); \
      |                 ^~~~~~~~~
glue/gluedebug.h:77:31: note: in expansion of macro 'doprint'
   77 | #define os_printf(x...)  do { doprint(x); } while (0)
      |                               ^~~~~~~
glue-lwip/arch/cc.h:76:36: note: in expansion of macro 'os_printf'
   76 | #define LWIP_PLATFORM_DIAG(x) do { os_printf x; } while(0)
      |                                    ^~~~~~~~~
lwip2-src/src/include/lwip/debug.h:150:34: note: in expansion of macro 'LWIP_PLATFORM_DIAG'
  150 |                                  LWIP_PLATFORM_DIAG(message); \
      |                                  ^~~~~~~~~~~~~~~~~~
glue-lwip/esp-ping.c:148:7: note: in expansion of macro 'LWIP_DEBUGF'
  148 |       LWIP_DEBUGF( PING_DEBUG, (" %"U32_F" ms\n", (sys_now()-ping_time)));
      |       ^~~~~~~~~~~
Multiple include guards may be useful for:/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/machine/_endian.hglue-lwip/lwip-err-t.h
```

The struct `ipv4_addr` having incomplete type in `user_interface.h` on lines [363](https://github.com/esp8266/Arduino/blob/1a381477ec47f12cda305b28ac70d3ea233a5fe2/tools/sdk/include/user_interface.h#L363)-548 when compiling `doprint.c` is remedied by removing `#include "esp-missing.h"` 
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/7421258237b7c8f61629226961af498a0a6e0096/glue/doprint.c#L45
When removed, it successfully compiles. The reason this compilation error occurs is because `esp-missing.h` includes `ipv4_addr.h`. The header file `ipv4_addr.h` includes `lwip/init.h` before it ever defines the struct `ipv4_addr`, and a header dependency (within the tree) of `lwip/init.h` turns out to be `user_interface.h` which tries to include `ipv4_addr.h`. As a result the struct `ipv4_addr` (technically compiler symbol) never gets defined until after its used in `user_interface.h`. Since I am currently unable to compile anything else, I do not know the ramifications of removing `#include "esp-missing.h"` in `user_interface.h` (not that it's this repo's responsibility to fix it as it's the [esp8266](https://github.com/esp8266/Arduino) repo.

Any more guidance would be appreciated for resolving the expansion errors.

d-a-v, 2021-05-12:
Something was indeed missing in xtensa-gcc-10.2
Can you try with latest master ?

Are52Tap, 2021-05-12:
@d-a-v That definitely did the trick regarding the expansion error, and with that one modification I mentioned (removing `#include "esp-missing.h"` from `doprint.c`) everything compiles successfully! Without removing that include, I get that error mentioned previously where `ipv4_addr` is an incomplete type, and does not compile.

Thank you very much, for your help so far. I'm willing to help you resolve this last issue I'm experiencing.

d-a-v, 2021-05-12:
Are you using this repo with esp8266 Arduino or with another project ?

Are52Tap, 2021-05-12:
@d-a-v I am doing this with `esp8266/Arduino`. I just now did a clean install of everything  (esp8266) as per your instructions [here](https://github.com/d-a-v/esp82xx-nonos-linklayer/issues/41#issuecomment-572556704). I modified `glue/gluedebug.h` to do lwip debugging as in this [patch](https://github.com/d-a-v/esp82xx-nonos-linklayer/files/6468943/lwipdebugging.txt), and then ran `make install` which gives those incomplete type errors as exactly shown [here](https://github.com/d-a-v/esp82xx-nonos-linklayer/files/6468947/lwipdebuggingerror.txt).

d-a-v, 2021-05-12:
I could reproduce.
I will fix that but if you change only `ULWIPDEBUG` it works.
`UDEBUG` is just about the glue, that's not really lwIP.

d-a-v, 2021-05-12:
You also need to update the lines [following these](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/66355f6958cee3ba4461fda50b56f426789120e1/glue-lwip/arduino/lwipopts.h#L3241-L3243) to enable what's needed for your investigations (then `make install`).

Are52Tap, 2021-05-12:
Awesome, it works! I should have payed more attention to the `UDEBUG` comments and tried messing around with those settings to see what would happen. It wasn't probably necessary to modify `UDEBUGINDEX` or `UDEBUGSTORE` for my purposes.
Thanks for the help. I'll let you close the issue.

d-a-v, 2021-05-12:
I'll close this issue after fix with `UDEBUG` enabled. Thanks for reporting !

Are52Tap, 2021-05-13:
@d-a-v I found another compilation issue caused by enabling `ULWIPDEBUG`. With it enabled in `tools/sdk/lwip2/include/gluedebug.h` and having `DEBUG_ESP_CORE` enabled (`Debug Level`: `Core` in Arduino IDE) it causes `cores/esp8266/LwipIntf.cpp` and `cores/esp8266/LwipIntfCB.cpp` fail to compile because `#include <stdio.h>` is no longer included from line [82](https://github.com/esp8266/Arduino/blob/1a381477ec47f12cda305b28ac70d3ea233a5fe2/tools/sdk/lwip2/include/lwip/arch.h#L82) (or [94](https://github.com/esp8266/Arduino/blob/1a381477ec47f12cda305b28ac70d3ea233a5fe2/tools/sdk/lwip2/include/lwip/arch.h#L94)) causing: 
```
/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/LwipIntf.cpp: In member function 'bool LwipIntf::hostname(const char*)':
/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/debug.h:8:28: error: '::printf' has not been declared
    8 | #define DEBUGV(fmt, ...) ::printf((PGM_P)PSTR(fmt), ## __VA_ARGS__)
      |                            ^~~~~~
/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/LwipIntf.cpp:114:9: note: in expansion of macro 'DEBUGV'
  114 |         DEBUGV("WiFi.(set)hostname(): empty or large(>32) name\n");
      |         ^~~~~~
/mnt/ext/arduino-1.8.13/hardware/esp8266com/esp8266/cores/esp8266/LwipIntf.cpp:17:1: note: 'printf' is defined in header '<cstdio>'; did you forget to '#include <cstdio>'?
   16 | #include "LwipIntf.h"
  +++ |+#include <cstdio>
   17 | 
```
Possible [patch 1](https://github.com/d-a-v/esp82xx-nonos-linklayer/files/6469992/patch.txt)
Possible [patch 2](https://github.com/d-a-v/esp82xx-nonos-linklayer/files/6469997/patch.txt)

Hopefully I don't keep finding issues.

d-a-v, 2021-05-14:
Your path1 would have to go into `patches/` because it's lwIP's official sources.
Your patch 2 or 3 would have to be submitted to esp8266 Arduino.
Will you submit a PR there ?