d-a-v/esp82xx-nonos-linklayer #63: ALTCP support

HamzaHajeir, 2023-05-31:
Hi there
LWIP offers [ALTCP APIs (Application Layered TCP)](https://www.nongnu.org/lwip/2_1_x/group__altcp__api.html) which offers a seamless shift into TLS support with their ALTCP_TLS.

The corresponding header files cannot be found by the compiler.

I use Arduino core.

What can I do to enable ALTCP?

With thanks,
Hamza Hajeir

mcspr, 2023-06-16:
See our build instructions, makefiles & cmake

In both cases, .c files can be omitted from builder and .h are only partially installed to esp8266/Arduino include dirs. For the first case, you would have to make changes to file lists in [makefiles/Makefile.lwip2](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/8e3c4eec00ec7f3962705ee095a310642953b5fb/makefiles/Makefile.lwip2) or [lwip-builder.cmake](https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/8e3c4eec00ec7f3962705ee095a310642953b5fb/cmake/lwip-builder.cmake)

iirc header installation is manual in makefile case, where d-a-v would filter out unused stuff like apps, sockets, altcp, etc. before doing a commit at esp8266/Arduino. Should probably install everything to avoid further issues, since I just noticed we have a bug with mismatching struct layout when using auto-ip feature because of that >_<

HamzaHajeir, 2023-06-19:
I see, altcp_* is excluded from the build.

Lwip master branch holds bugfixes to ALTCP, in my seek to test on ESP32
I've manually cherry-picked the commits I'm concerned about, then sourced
as a custom esp-idf component.

But what's the optimal way to do a similar solution here?

And for other headers and sources, isn't that correct they won't increase
the application binary size by the linker?

I know it's better to be explicit about what we need. But I'm interested in
supporting TLS with the Arduino core natively (no custom build) to the
library users (H4AsyncTCP).

mcspr, 2023-06-19:
Note of the version difference, most likely upstream v2.1.3 already has those
For cherry picking anything else, add it as patches/*.patch?

To modify build flags for the lib
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/master/glue-lwip/arduino/lwipopts.h

Unused functions are ignored in the resulting build

HamzaHajeir, 2023-06-20:
There's about 18 related commits between 2.1.3 release and master, really.

Good hint for patches applying.

After a successful build and test, do you think that can be added to the
Arduino core?

HamzaHajeir, 2023-06-22:
> No reason why not; isn't the whole idea of altcp is just extra api on top
> of existing one? so, if we internally or otherwise don't use it, nothing
> changes or raw api callers start to behave differently

I see and I agree with you.

A side discussion: I'm seeing less activity for the Arduino core of 8266,
why do you think?

d-a-v, 2023-07-11:
> A side discussion: I'm seeing less activity for the Arduino core of 8266,

Because it's just working ?

I personally was and still am busy with other project.
As @mcspr said, no reason why it couldn't be integrated (header files and linkable code in library).
Did altcp work in your test ?

HamzaHajeir, 2023-07-12:
Hi David,

Yes it does work for the ESP32 target. Didn't progress for the 8266.

One can try easily with this environment:
https://github.com/HamzaHajeir/H4Plugins_Env

BTW, these are the provided patches regarding ALTCP_TLS over LwIP master:
https://savannah.nongnu.org/patch/?10368

d-a-v, 2023-07-18:
Is ALTCP requiring MbedTLS ?
We don't have it yet on esp8266.
We are using BearSSL instead. Would ALTCP be able to use something else than MbedTLS ?

HamzaHajeir, 2023-07-18:
Yes, the available is mbedTLS port.

Although ALTCP is designed as virtual layer, but it requires separate
development.

d-a-v, 2023-07-18:
Supporting MbedTLS on esp8266 Arduino is quite an amount of work.
Are you expecting to use this repository with something else on esp8266 other than the arduino core ?

HamzaHajeir, 2023-07-19:
I see the point,

I've no expectations other than Arduino for the esp8266.

HamzaHajeir, 2023-08-05:
Hi

I think if BearSSL API is similar to mbedTLS API regarding SSL, there's a possibility to copy and tweak `altcp_tls_mbedtls.c`.

d-a-v, 2023-08-07:
OK then,
if you're using makefiles to build lwip2, you can try with this patch.
You would have to create a `lwip2-src/src/apps/altcp_tls2/` (or whatever name you wish to use) and put your tweaks in there.

```diff
diff --git a/glue-lwip/arduino/lwipopts.h b/glue-lwip/arduino/lwipopts.h
index bcbda45..1f43e5e 100644
--- a/glue-lwip/arduino/lwipopts.h
+++ b/glue-lwip/arduino/lwipopts.h
@@ -1499,7 +1499,7 @@
  * See @ref altcp_api
  */
 #if !defined LWIP_ALTCP || defined __DOXYGEN__
-#define LWIP_ALTCP                      0
+#define LWIP_ALTCP                      1
 #endif
 
 /** LWIP_ALTCP_TLS==1: enable TLS support for altcp API.
@@ -1510,6 +1510,9 @@
 #if !defined LWIP_ALTCP_TLS || defined __DOXYGEN__
 #define LWIP_ALTCP_TLS                  0
 #endif
+#if !defined LWIP_ALTCP_TLS2 || defined __DOXYGEN__
+#define LWIP_ALTCP_TLS2                 1
+#endif
 
 /**
  * @}
diff --git a/makefiles/Makefile.lwip2 b/makefiles/Makefile.lwip2
index 6a4ccbd..1e504e1 100644
--- a/makefiles/Makefile.lwip2
+++ b/makefiles/Makefile.lwip2
@@ -10,6 +10,8 @@ OBJ = \
        $(patsubst %.c,$(BUILD)/%.o,$(wildcard api/*.c)) \
        $(patsubst %.c,$(BUILD)/%.o,$(wildcard apps/sntp/*.c)) \
        $(patsubst %.c,$(BUILD)/%.o,$(wildcard apps/mdns/*.c)) \
+ $(patsubst %.c,$(BUILD)/%.o,$(wildcard apps/altcp_tls2/*.c)) \
+ $(patsubst %.c,$(BUILD)/%.o,$(wildcard apps/http/*.c)) \
 #      $(subst ../../lwip2-contrib-src/,contrib/, \
 #              $(patsubst %.c,$(BUILD)/%.o,$(wildcard ../../lwip2-contrib-src/apps/ping/*.c)))
```

mcspr, 2023-08-07:
`LWIP_ALTCP_TLS_MBEDTLS`
`LWIP_ALTCP_TLS_BEARSSL`
?

I think the idea is `LWIP_ALTCP_TLS` enables TLS, another flag selects implementation. Plus some code to `#error` out when building with both

d-a-v, 2023-08-07:
This one is defined and used in in  `altcp_tls_mbedtls*`
It should be indeed added (or `LWIP_ALTCP_TLS_BEARSSL`) in `lwipopts.h`

```c
lwip2-src/src/include/lwip/apps/altcp_tls_mbedtls_opts.h:#ifndef LWIP_ALTCP_TLS_MBEDTLS
lwip2-src/src/include/lwip/apps/altcp_tls_mbedtls_opts.h:#define LWIP_ALTCP_TLS_MBEDTLS                        0
```

mcspr, 2023-08-07:
Don't forget that bssl also has some changes so we don't trigger watchdog timer

```
~/.p/p/f/t/s/s/bearssl ((b024386d))> git grep 'yield('
src/ec/ec_prime_i15.c:  optimistic_yield(10000);
src/ec/ec_prime_i31.c:        optimistic_yield(10000);
src/inner.h:  extern void optimistic_yield(uint32_t);
src/inner.h:  #define optimistic_yield(ignored)
src/rsa/rsa_i15_priv.c: optimistic_yield(10000);
src/rsa/rsa_i15_priv.c: optimistic_yield(10000);
src/rsa/rsa_i15_priv.c: optimistic_yield(10000);
src/rsa/rsa_i31_priv.c: optimistic_yield(10000);
src/rsa/rsa_i31_priv.c: optimistic_yield(10000);
src/rsa/rsa_i31_priv.c: optimistic_yield(10000);
```

Besides modifying our mbedtls port in a similar way, there are `MBEDTLS_*_ALT` that are sort-of the thing we might be looking for to avoid the issue.
(doc does not seem to mention anything when searching for either yield or context switching related keywords)